danak6jq/Seeed-LoRa-E5 #1: Add example or tutorial

simon88, 2021-12-13:
Hi,

Nice work it's not an issue but it would be nice to add examples how to use your repo step by step with Lora-e5

danak6jq, 2021-12-13:
Thank you! I've added a very simple README that assumes the user is familiar with STM32CubeIDE and LoRaWAN concepts. I trust it's better than nothing; this repo isn't intended as a beginner's tutorial, but I'm happy to enhance the README, especially if offered PRs 💯

danak6jq, 2021-12-13:
BTW the short answer, for anyone familiar with LoRaWAN and STM32CubeIDE, is literally "clone the repo, open project from filesystem, you know the rest"  :-)

simon88, 2021-12-13:
Yep thanks the README is perfect :). BTW right now I can't erase AT firmware I try to Open STM32CubeProgrammer, connect ST-LINK, hold RESET Button, then click Connect and release RESET Button but I got `No STM32 device found!` I'll looking for that this is strange, thanks again for your repo

danak6jq, 2021-12-13:
@simon88 it's been a while since I erased the LoRa-E5 and I don't recall the details. I have the ST/Link connected to the reset pin - have you tried that?

danak6jq, 2021-12-13:
@simon88 if you're still scratching your head, I'll have a look at another LoRa-E5 Mini I have that hasn't been erased yet. Let me know

simon88, 2021-12-13:
@danak6jq I've connected my st-link to the board like it explained in the Seed wiki here https://wiki.seeedstudio.com/LoRa_E5_Dev_Board/) GND, SWDIO and SWCLK so for you I've to connect also NRST pin? I've updated st-link firmware also.

danak6jq, 2021-12-13:
Yeah, connect the nRST pin and see if that helps (bet it will)

simon88, 2021-12-14:
@danak6jq same issue for me with nRST pin connected to my board, I really don't understand why it's doesn't works for now I have this message `Error: ST-LINK error (DEV_TARGET_CMD_ERR`  it's frustatigng beceause your code works well and I blocked with Seed board

danak6jq, 2021-12-14:
Ah yes, ST/Link V2 wants to see the target Vcc. Excellent!

simon88, 2021-12-15:
Just a question, I'm playing with your code and I've a question, for example for USART1 is it normal thaht I havent a auto generated code for this ? When I regenerate the code, in the mian.c after in ` Initialize all configured peripherals` section I don't see any mx function to initialize usart1, it's works well for example with i2c

danak6jq, 2021-12-15:
@simon88 that's normal - the structure of the application produced by Cube here has the USART initialization down in the trace initialization in usart_if.c. One of the quirks of the project is that most of the peripheral initialization is not done directly from main() - I learned this when building the project from scratch.

closed 2021-12-15 by danak6jq

RocketManRC, 2022-01-10:
@danak6jq I don't know if I should be commenting here since it is closed but first of all I want to say that your work to put this repo together and all the comments in the issues has really helped me a lot and I really appreciate it.

I also had a problem with my STLink not being recognized. I had both the Dev Board and the STLink plugged into a USB hub and when I moved the STLink to a dedicated USB port all was well. I'm doing this on MacOS so it may be different behavior than for Win10 but I thought I would mention it.

I will try connecting VCC to the STLink and then try connecting it to the hub again as that will be a lot more convenient.

Cheers!

danak6jq, 2022-01-10:
@RocketManRC you're quite welcome, I am happy to help. No problem commenting in a closed question issue like this, especially when it's a thank-you note LOL. I'm not keeping a hints/tricks page in the repo yet, but you're certainly welcome PR such a thing.

My set-up is on a solderless breadboard with external power; I realized that connecting power to the "+" and "-" terminals on the PCB bypasses the USB-serial converter and powers the MCU through an LDO (also the 3v3 pin on the board). This way the USB-serial converter is powered-off, and the only use is the LDO and MCU; in low-power mode, I measure 90uA. When the MCU wakes up, it uses several mA, and when it transmits, it spikes higher, depending on TX power setting.

By using this, I've already got a battery-powered mote outside with an SHTC3 therm/hygro. Sampling the temperature every 10 seconds and sending a report when either the temperature or humidity change enough (0.5C or 0.5%). I bet the battery will last weeks.

danak6jq, 2022-01-10:
@simon88 one thing I noted that you probably already have - if you enable low-power mode in the LoRaWAN application settings and configure additional peripherals (like, in my case, I2C), the peripheral loses power while powered-down. So, before using it, you need to re-initialize it, I just called MX_I2C2_Init() from the send TX data function and that does the trick. I suspect other peripherals will require something similar

RocketManRC, 2022-01-11:
@danak6jq Thanks for your reply. I would be happy to start on a "hints/tricks" page and send a PR. Let me see what I can do to get it started :-)

I became interested in the LoRa-E5 because I wanted to experiment with making a tracker using the Helium Network. So far I have had great success with the Grove version and an external micro as controller using the AT interface but now it looks like I can do everything right on the Dev Board. Next for me is to do the same with the Grove board itself and then add GPS to it. Also I am also very interested in your experiments with  low power modes and have done some work on that as well.

I am keen to collaborate on any or all of this!

danak6jq, 2022-01-11:
@RocketManRC I've simply enabled the Low Power mode in the LoRaWAN application settings; I haven't confirmed which sleep/power-down mode it's using, but it's evident that memory is retained, the 32kHz clock + timers run, and some peripherals require re-initialization upon waking. Really easy so far, and I've measured actually 69uA current consumption in that mode, which I'm pretty happy with.

RocketManRC, 2022-01-11:
@danak6jq  Oh wow that completely went over my head the first time you mentioned it. I will try it out tomorrow! Yes 69uA will be perfect for what I am trying to do.

polihedron, 2022-03-17:
@danak6jq,
I managed to measure current consumption in stop2 mode low as in datasheet (2uA),
when I remove 3V3 LDO regulator, and power LoRa-E5-mini module via 3V pin with CR17450 3V battery:
[image]

[image]

Kind Regards

arvidblaser, 2022-03-28:
I just wanted to also add an extra thank you post here, both for the repo that does what seeed studio kind of failed with, and for the comment about the voltage regulator. It really works perfect programming a real low power application now with an average of 1.9uA in sleep^^. 

If you reconfigure the board to use i2c3 instead of i2c2 you also won't need to reinit i2c after sleep also (however that changes the pins used so it maybe won't suit everybody to do so).