danak6jq/Seeed-LoRa-E5 #4: IN865 - downlink not working in Class-A (MAC rxTimeOut)

punee995, 2021-12-20:
First of all thanks for the clean example.

I'm trying to do send downlink on Port 2(for led) but when I send downlink from chirpstack I'm getting **IRQ_RX_TX_TIMEOUT** can you give me some hint where to look ?

Also funny thing is In Class-C downlink works properly.

```APP_VERSION:        V1.1.0
MW_LORAWAN_VERSION: V2.3.0
MW_RADIO_VERSION:   V1.1.0
###### OTAA ######
###### AppKey:      8B:0C:9A:82:E6:74:A5:11:B9:E5:26:A9:11:87:EB:84
###### NwkKey:      8B:0C:9A:82:E6:74:A5:11:B9:E5:26:A9:11:87:EB:84
###### ABP  ######
###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEui:  2C:F7:F1:20:24:90:0E:A5
###### AppEui:  01:01:01:01:01:01:01:01
###### DevAddr: 05:03:EF:F5
1640001408s476:TX on freq 865402500 Hz at DR 0
1640001409s981:MAC txDone
1640001415s005:RX_1 on freq 865402500 Hz at DR 0
1640001416s824:MAC rxDone

###### = JOINED = OTAA =====================

###### ========== MCPS-Indication ==========
1640001418s499:temp= 21
1640001418s499:VDDA= 254
1640001418s503:TX on freq 865062500 Hz at DR 0
1640001418s516:SEND REQUEST
1640001420s162:MAC txDone
1640001421s187:RX_1 on freq 865062500 Hz at DR 0
1640001422s349:MAC rxDone
1640001422s352:VDDA= 254

###### ========== MCPS-Confirm =============

###### ========== MCPS-Indication ==========
1640001428s499:temp= 21
1640001428s499:VDDA= 254
1640001428s503:TX on freq 865985000 Hz at DR 0
1640001428s516:SEND REQUEST
1640001430s326:MAC txDone
1640001431s351:RX_1 on freq 865985000 Hz at DR 0
1640001432s513:MAC rxDone

###### ========== MCPS-Confirm =============

###### ========== MCPS-Indication ==========
1640001438s499:temp= 21
1640001438s499:VDDA= 254
1640001438s503:TX on freq 865402500 Hz at DR 0
1640001438s516:SEND REQUEST
1640001440s162:MAC txDone
1640001441s187:RX_1 on freq 865402500 Hz at DR 0
1640001442s349:MAC rxDone

###### ========== MCPS-Confirm =============

###### ========== MCPS-Indication ==========
1640001448s499:temp= 20
1640001448s499:VDDA= 254
1640001448s503:TX on freq 865985000 Hz at DR 0
1640001448s516:SEND REQUEST
1640001450s162:MAC txDone
1640001451s187:RX_1 on freq 865985000 Hz at DR 0
1640001452s349:MAC rxDone

###### ========== MCPS-Confirm =============

###### ========== MCPS-Indication ==========
1640001458s499:temp= 21
1640001458s499:VDDA= 254
1640001458s503:TX on freq 865062500 Hz at DR 0
1640001458s516:SEND REQUEST
1640001460s162:MAC txDone
1640001461s187:RX_1 on freq 865062500 Hz at DR 0
1640001461s394:IRQ_RX_TX_TIMEOUT
1640001461s394:MAC rxTimeOut
1640001462s150:RX_2 on freq 866550000 Hz at DR 2
1640001462s226:IRQ_RX_TX_TIMEOUT
1640001462s226:MAC rxTimeOut
1640001465s115:TX on freq 865402500 Hz at DR 0
1640001466s773:MAC txDone
1640001467s797:RX_1 on freq 865402500 Hz at DR 0
1640001468s499:temp= 22
1640001468s499:VDDA= 254
1640001468s959:MAC rxDone
```

danak6jq, 2021-12-20:
I believe this just means  the gateway is not sending the downlink in the receive window. Class C leaves the receiver running all the time, so this isn't a problem.

Class A only listens for a downlink in precise time windows after a UL, so I'd check to make sure the LoRa network server and gateway are configured correctly. This is most likely a misconfiguration of the server and gateway.

What kind of gateway and server are you using?

danak6jq, 2021-12-20:
I just re-built and test DL for LED control, port 2, and verified it works with my ChirpStack server. Odds are there is an issue with the server configuration there, also possibly with IN865 regional parameters.

punee995, 2021-12-21:
Thanks @danak6jq, Downlink is working now, and yes It was because of regional parameters.
[en.STM32WL-Peripheral-LoRaStack_LORASTACK.pdf](https://www.st.com/content/ccc/resource/training/technical/product_training/group1/ee/e6/ef/5b/7b/43/48/95/STM32WL-Peripheral-LoRaStack_LORASTACK/files/STM32WL-Peripheral-LoRaStack_LORASTACK.pdf/_jcr_content/translations/en.STM32WL-Peripheral-LoRaStack_LORASTACK.pdf)

closed 2021-12-21 by punee995