danak6jq/Seeed-LoRa-E5 #5: SX126x SPI to ARM

ramimehyar, 2022-01-10:
Can you please tell us if you can in addition to SPI, on what other pins to the MCU is the SX126x integrated, Busy and Serial select.... etc?

danak6jq, 2022-01-10:
Are you asking how the STM32WLxx integrates the SX126x radio internally? If so, I'd point at the ST documentation.
If you're asking which GPIOs are used for RF switching, that's Seeed LoRa-E5 documentation item.

Are you asking something else?

ramimehyar, 2022-01-10:
Ho Dana,

You had my questions correct, unfortunately, on both documentation it is
only mentioned that they are integrated through SPI, and the rest of the
ports are not defined.

If i want to develop my own application, i will need to configure ot
accordingly!

Maybe i am missing something, how can i use the RF module without the AT
command? Can you point me in that direction at least please?

Regards,

Rami

danak6jq, 2022-01-10:
This repo was produced using STM32CubeIDE to assemble ST's firmware support into a project. It started as a functional clone of the LoRaWAN_End_Node example, but it is a stand-alone project. If you clone this repo and open it with STM32CubeIDE, build it, and flash it into a LoRa-E5 , you will be using the RF module without the AT firmware, this replaces it. 

You do not need to do a thing to configure the LoRaWAN radio or stack - it's all integrated in this project.

The LoRaWAN application is contained in lora_app.c 

To build a new LoRaWAN application (mote), you'd probably start by changing code in lora_app.c - because this is the ST example code, it is as instructive as I can be.

closed 2022-01-10 by danak6jq

ramimehyar, 2022-01-10:
Thank you Dana, that helps alot!

I will give it a try.. However, i am planning to develop for the nodes to
send directly to each other, i.e Lora and no Lorawan! I guess i need to
research further to find an example where i can send from one node to
another directly.

Regards,

Rami

danak6jq, 2022-01-10:
For point-to-point communication examples, have a look at the example SubGHz_Phy_PingPong in the STM32WLxx firmware package. You can easily create it as a STM32CubeIDE project (create new project from an example) and it sounds like what you want.

ramimehyar, 2022-01-10:
Awesome, thanks!