danak6jq/Seeed-LoRa-E5 #7: Issue building project

scott1423, 2022-02-13:
Hi
  I have the latest ST Cube MX, IDE etc, and having issues with building the project.

I done the following:
 1. Copied the code from here into my working folder.
 2. Tried opening the project in the IDE, but nothing there.
 3. I opened the project in the cubeMX and created the project, then tried opening project, but again no project within the IDE was displayed.
 
Do I have to load a project in first, before loading (overwriting ) files from this page?

Many Thanks
Scott

scott1423, 2022-02-13:
Hi,
  Thanks for the reply.

1. I downloaded your source code to my working folder.
2. I ran the IDE and selected the project, as follows:
[image]
3. I selected the folder as follows:
[image]
4. Then pressed finish.
5. Nothing appears , as follows:
[image]
6. I then opened it up in cubeMX and then Generated project. All seem to be generated going by cubeMX screen. I selected open project but got no project.

Not sure what I am doing wrong :(

Many thanks
Scott

danak6jq, 2022-02-13:
I created C:\Work, downloaded the repo as a zip-file, extracted to C:\Work

[image]

Then opened the 'Projects from filesystem' dialog:

[image]

Which makes me think the correct directory to import from is:

[image]

Which imports this project:

[image]

STM32CubeIDE is pointing out the project's name is really Seeed-LoRa-E5 even though the directory has '-master' appended.
But it did import the project, which builds just fine.

Which version of STM32CubeIDE are you using?

[image]

I've updated mine to v1.8.0 but I believe v1.7.0 will work.

Here, I've renamed the folder to remove -master before importing:

[image]

and STM32CubeIDE doesn't mention the directory name now:

[image]

and this builds as expected:

[image]

scott1423, 2022-02-13:
Hi
  I cleaned out my work folder, and followed what you had above, and works now.  I am also using the same version of IDE as you.  I entered my app and network key, but doesnt seam to join, see below log file. Any idea why this is? 

Many thanks
Scott

APP_VERSION:        V1.1.0
MW_LORAWAN_VERSION: V2.3.0
MW_RADIO_VERSION:   V1.1.0
###### OTAA ######
###### AppKey:      8B:0C:9A:82:E6:74:A5:11:B9:E5:26:A9:11:87:EB:84
###### NwkKey:      3A:09:42:6C:23:0B:F8:83:40:B0:40:24:1C:DD:47:C3
###### ABP  ######
###### AppSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### NwkSKey:     2B:7E:15:16:28:AE:D2:A6:AB:F7:15:88:09:CF:4F:3C
###### DevEui:  2C:F7:F1:20:32:30:52:2C
###### AppEui:  80:00:00:00:00:00:00:06
###### DevAddr: 00:17:FE:06
0s040:TX on freq 868500000 Hz at DR 0

[2022-02-13 18:52:06.732]# RECV ASCII>
1s527:MAC txDone

[2022-02-13 18:52:11.764]# RECV ASCII>
6s561:RX_1 on freq 868500000 Hz at DR 0

[2022-02-13 18:52:11.971]# RECV ASCII>
6s759:IRQ_RX_TX_TIMEOUT
6s759:MAC rxTimeOut

[2022-02-13 18:52:12.768]# RECV ASCII>
7s561:RX_2 on freq 869525000 Hz at DR 0

[2022-02-13 18:52:12.970]# RECV ASCII>
7s759:IRQ_RX_TX_TIMEOUT
7s759:MAC rxTimeOut

###### = JOIN FAILED

###### = RE-TRYING OTAA JOIN
7s765:TX on freq 868100000 Hz at DR 0

[2022-02-13 18:52:14.455]# RECV ASCII>
9s252:MAC txDone

[2022-02-13 18:52:15.251]# RECV ASCII>
10s044:temp= 22

10s044:VDDA= 254

[2022-02-13 18:52:19.501]# RECV ASCII>
14s287:RX_1 on freq 868100000 Hz at DR 0

[2022-02-13 18:52:19.685]# RECV ASCII>
14s485:IRQ_RX_TX_TIMEOUT
14s485:MAC rxTimeOut

danak6jq, 2022-02-13:
What do you see on your server logs for your gateway? Do you see the join requests? I see you're EU868, right?

scott1423, 2022-02-13:
Hi,
  Yes, I am on 868MHz.  It is a bit confusing, especially as I am new to Lora :)

Looking on the End Device screen I have it set as following:
[image]

[image]

And on the Gateway screen I have:

[image]

closed 2022-02-13 by danak6jq

scott1423, 2022-02-14:
> Glad I could help! Enjoy.

Many thanks again for all your help. Im now getting packets of data to TTN server ..  I was wondering if you have managed to be able to set commands within the Lora E5? For eample, how would I set transmition channel, and change the spreading factor. If you have any idea, would you mind passing this on?

Many thanks again

Scott

danak6jq, 2022-02-14:
The STM32WL in the LoRa-E5 is well-supported by ST firmware and documentation, so you'll want to look at the ST site. For example,  though the naming of this seems odd, this manual describes the APIs to the LoRaWAN layer in section 4.6 and 4.7:

https://www.st.com/resource/en/user_manual/dm00300436-stm32-lora-expansion-package-for-stm32cube-stmicroelectronics.pdf

For example, to set the transmit Data Rate (which determines spreading factor), you'd use this function:

[image]

Note that when ADR is enabled, the network server may change DR as it sees fit.
Also, LoRaWAN protocol automatically changes channels for Uplinks.

danak6jq, 2022-02-14:
Have a look at Table 34 in section 4.8:

[image]

Lots of cool functions there.

Generally, though, ADR enable/disable is done at initialization time:

[image]

[image]

that's in lora_app.c

scott1423, 2022-02-14:
Many thanks again... Very much appreciated, again. You have been a great help.

Best Regards
Scott

danak6jq, 2022-02-14:
You're very welcome, @scott1423

scott1423, 2022-03-21:
Hi
  I have been using your project a while, and all working great. I was wondering if you can give a bit of advice? I wish to have 1 device as a server, and other nodes sending lora packets to it. Any idea the best way to go here?

Many thanks again
Scott