danak6jq/Seeed-LoRa-E5 #8: Update for v1.2.0 of STM32Cube MCU Package for STM32WL series(SDK)

polihedron, 2022-03-17:
Hi there, not an issue,
can you update project for new v1.2.0 STM32WL SDK please?
https://www.st.com/content/my_st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32cube-mcu-mpu-packages/stm32cubewl.license=1608693595598.product=STM32CubeWL.version=1.2.0.html#overview

danak6jq, 2022-03-17:
I've been looking-forward to the release of V1.2.0 STM32WL - will do. Thank you!

polihedron, 2022-03-18:
that's great :)

danak6jq, 2022-03-19:
@polihedron I did a quick merge of V1.2 as well as some clean-up of the BSP support for LEDs/buttons removed in the V1.2 example. Builds and runs on my LoRa-E5 Mini without issue; pretty sure I got the RF switch bits correct. Please open an issue if there's a problem.

closed 2022-03-19 by danak6jq