danak6jq/Seeed-LoRa-E5 #11: P2P application with LoRa-E5 mini

hpernet, 2022-07-07:
Hello, 
I'm currently trying to make a P2Pproject using SubGHz with a LoRa-E5 (like the pingpong exemple from ST)
But when I try to replace LoRaWAN layer by a ping pong application, SUBGHZSPI_Transmit return HAL_SUBGHZ_ERROR_TIMEOUT.
And when I try to start a new project from your ioc file, radio events never occured (OnTxDone onRxDone...).
Do you have more informations, or tips to make a P2P application with a LoRa-E5 mini ? 

Thanks 

Hugo

danak6jq, 2022-07-07:
I've never built the P2P sample app from ST or looked at building a non-LoRaWAN application for the STM32WLxx. I'm guessing you can do basically what I did here - create a new Cube project, configure all the peripherals, and then manually paste in the ST example code into the resulting framework. You may also make use of the Drivers/BSP code from this repo. 

But the P2P app is outside the scope of this repo - you're looking for help here. Have you tried the ST forum?

hpernet, 2022-07-07:
Thanks for your anwers.
Yes I'tried St forum, seeds forum, but it's look like nobody succed to do this...
Do you know if there is a specific configuration to enable the radio output ?

hpernet, 2022-07-08:
After mor trial, I think that my problem is more general...
I try your programme without modification (only comment the debug init function to be able to debug the target)
And I also enter into the timeout error in SUBGHZSPI_Transmit function... It seams like TXE flag is never set.

Do you have a particular hardware configuration ? Option Byte config ? 
Thanks

danak6jq, 2022-07-08:
You might disable low-power mode in the Cube configuration to see if there's an issue with debugging and low-power. Otherwise, I have no idea; this repo just works on Seeed LoRa-E5s here

hpernet, 2022-07-12:
I will try that thanks for your help