danak6jq/Seeed-LoRa-E5 #12: TTN/HeliumConsole Payload Decoder

Saasi-FPV, 2022-07-08:
I'm working on a GPS Tracker based on a LoRa-E5.
Now I'm facing a problem... I can't figure out how to decode the Payload. I found the section in code where the formatting is done (lora_app.c) but reversing it gives me a lot of trouble.
Is out there in the WWW a Decoder that I overlooked? Or maybe you have one at hand?

I thank you for the great work you've done with the "Clean" file! Seed should use your example! :)

danak6jq, 2022-07-08:
The decoder is in your LoRaWAN network server, right? Which server are you using?

Saasi-FPV, 2022-07-10:
Yes. I'm using the Helium Console. But it shouldn't matter they are always written in JavaScript. 
While searching around I found this: https://github.com/helium/console-decoders/blob/master/SEEED/Seeeduino_LoRaWAN/decoder.js but this one is for another project.

closed 2022-07-10 by danak6jq

danak6jq, 2022-07-10:
I don't know how to help with this - it certainly is outside the scope of this repo. Good luck!