danak6jq/Seeed-LoRa-E5 #27: Modifying Pinout - IOC Code Regeneration Question

KonRoz, 2024-03-22:
Building with the default configuration works fine. That said, when changing the pinout in CubeMX (e.g., enabling a GPIO) regenerating the code throws several errors. Am I missing something basic? I'm hoping to use the I2C and UART interfaces on the STM32WL in the LoRa-E5 module, but have had trouble getting over this hump. Thanks!

thonghatmit, 2024-05-24:
i have the same problem

thonghatmit, 2024-05-24:
for me, i need a ADC pin, but ioc generate faulty code

thonghatmit, 2024-05-26:
i think so, it's like they did not optimize for every code generator tool chain.

thonghatmit, 2024-05-26:
it's very complicate problem. With STM32CubeIDE tool chain, when i generate code with adc pin, adc_if.h is gone, i back up this file, but some code like generate DBG pin is gone, i think some configuration code is outside user code tag section

wql2002, 2024-11-27:
Same issue. I also wanted to add one pin to adc. After regenerating the code, I found that the code base had huge changes. So my solution is to find out the real difference of adc-related codes and manually change it to the original codebase instead of using .ioc to regenerate the project.